Esoterical/voron_canbus #62: Failure to manuconfig for Katapult

Jesusday, 2024-04-14:
When trying to get the make manuconfig under the installing katapult, it says there is not a file.

"pi@VoronTriZero:~/katapult $ make menuconfig
Makefile:139: out/board-link: No such file or directory
make: *** No rule to make target 'src/Kconfig', needed by '/home/pi/katapult/.config'.  Stop."

This is probably an easy fix for someone who knows Lenix but that is not me. 

I have a Pi 5 on this machine that is failing.
I have followed your steps on two older printers and they did work for them with older pi's. I tried copying the katapult folder from one of those pi to this pi 5 and it seems like there is not enough room for the folder and it will not copy everything over. 
I would assume that is why there is a missing folder where the command is looking for and causing the fails to build. 

Here are the commands leading up to this failure point:
"pi@VoronTriZero:~ $ sudo apt update
sudo apt upgrade
sudo apt install python3 python3-pip python3-can
pip3 install pyserial
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3 is already the newest version (3.11.2-1+b1).
python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
python3-can is already the newest version (4.1.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
pi@VoronTriZero:~ $ sudo test -e ~/katapult && (cd ~/katapult && git pull) || (cd ~ && git clone https://github.com/Arksine/katapult) ; cd ~
Already up to date.
pi@VoronTriZero:~ $ cd ~/katapult
pi@VoronTriZero:~/katapult $ make menuconfig
Makefile:139: out/board-link: No such file or directory
make: *** No rule to make target 'src/Kconfig', needed by '/home/pi/katapult/.config'.  Stop."

Esoterical, 2024-04-14:
Here is not a good place for support, but for your specfic issue I'd try removing your katapult folder
`rm -rf ~/katapult`
then getting a fresh version
`cd ~ && git clone https://github.com/Arksine/katapult`
then trying the `cd ~/katapult` and `make menuconfig` again.
If you still have issues the best place I am (and others) is the "can-and-usb_toolhead_boards" channel on the VORONDesign discord https://discord.com/channels/460117602945990666/1076243803947667516

Jesusday, 2024-04-14:
It worked! 
I guess deleting it through WinSCP and re-creating is not the same.

Thank you for your reply! It was not till after making this issue that I realized Katapult has its own github.

closed 2024-04-14 by Jesusday