Esoterical/voron_canbus #75: dfu-util 0.9 fails to upload to Octopus Pro F429

Graylag-PD, 2024-06-05:
I followed the guide with eventual success (great work, thank you!), but I found one stopper issue which is not mentioned in guide and would be a stopper for less experienced user. 
Older Debian releases come with dfu-util in version 0.9, which is however not able to flash Octopus Pro F429 (and possibly other boards).  At the point when it shoud say `Internal Flash => 0x08000000` instead it says just `UNKNOWN`. At this point it is game over as you cannot proceed. 

Solution is to update dfu-util to newest version 0.11, but this is unfortunately not straight forward. Using apt-get does not offer anything newer than 0.9. And if you try to follow build guide on dfu-util page (https://dfu-util.sourceforge.net/build.html), it fails as cloned repo does not have correct files (probably not an issue for experienced users, but a road block for "not-so-experienced" (just follow the guide) users). 
If you are stuck in same situation, just follow this (not saying it is the best approach, but it worked for me):
Download latest release tarbal `[dfu-util-0.11.tar.gz](https://dfu-util.sourceforge.net/releases/dfu-util-0.11.tar.gz)` and save it into your /home folder
Run `tar zxvf dfu-util-0.11.tar.gz` which will unpack the source code
After that just follow the guide on dfu-util page after the git clone line, so run:
`cd dfu-util
./autogen.sh
./configure
make
sudo make install`
After this, your dfu-util should have latest version and rest of the canbus guide can be followed.

Perhaps this should be included in the canbus guide or at least mentioned as possible issue. Regular klipper user without linux experience would have almost zero chance of getting through.

JoshuaKabo, 2025-01-20:
I ran into the same problem and found this very helpful! Also I had to reboot to get dfu-util -V to show up as 0.11 after the last step.

Graylag-PD, 2025-01-20:
Thanks, I hoped this will help at least a single person. I hate when I run into this kind of issues when it seems to be simple, yet there is no solution anywhere. Or worse, when you find a thread dealing with exactly your issue, which abruptly ends by OP simply stating "solved" without ever saying how