Esoterical/voron_canbus #83: Suggested update to "Getting Started" section  systemd-networkd (netplan)

clembo1, 2024-11-11:
The following link has some additional instructions on using Systemd/Network to configure CAN.  
https://www.pragmaticlinux.com/2021/07/automatically-bring-up-a-socketcan-interface-on-boot/

Background:

1. Just configuring 10-can.link and 25-can.network does not guarantee that the canbus will be started automatically.  I tested and can confirmed that nothing auto-started even post reboot.  Additional steps are required; see URL above for details.  
2. On Armbian - Debian and Ubuntu distros,  ifupdown and net-tools are not installed by default since Netplan is used by the distros.  This results in missing folders & files for /etc/network/.  While users should know linux, hunting down what packages are needed can be frustrating while to try following existing instructions for ifupdown section. Note:  Installing the packages is easy but this option may or may not be available in the future (due to deprecation) based on what I read.
3. You can use both NetworkManager (netplan) and systemd-networkd at the same time just as long as you don't try to control the same network device through both interfaces.  Debian, Ubuntu and Fedora install systemd-networkd by default. The installer just leaves it inactive.  

Instructions that worked for me section  systemd-networkd (netplan) | copied in part from the url
1.  Pre-req's -- you have already got your mainboard flashed.  CAN shows up in lsusb.  In my case it is:
Bus 002 Device 004: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter

2.  Complete the 10-can.link and 25-can.network configurations in the original instructions.
Note:  I added an additional line on 25-can.network to included RestartSec=100ms.
[Match]
Name=can*
[CAN]
BitRate=1M
RestartSec=100ms

3. Load the SocketCAN kernel modules
Before we can bring the can0 SocketCAN network interface in the UP state, we need to first load the SocketCAN related kernel modules. Open up the terminal again and run these commands:

    sudo modprobe can
    sudo modprobe can_raw
To verify that the SocketCAN related kernel modules loaded properly, you can run lsmod | grep "can".

Should get something like this:
voron@bigtreetech-cb1:~$ lsmod | grep "can"
can_dev                36864  1 gs_usb
can_raw                20480  0
can                    20480  1 can_raw

5. Configure and bring up the SocketCAN network interface

With the USB-to-CAN adapter plugged in and the SocketCAN kernel modules loaded, we continue with configuring the can0 SocketCAN network interface. The command for this is:

    `sudo ip link set can0 type can bitrate 1000000 restart-ms 100`

_Note: the default with current instructions does not set the restart timeout, not sure if 100 milliseconds is needed or not_ 
This configures the can0 network interface for a CAN communication speed of 1,000,000 bits/second. Feel free to adjust this to match the CAN communication speed on your CAN bus. Furthermore, it configures can0 to automatically re-initialize the CAN controller in case of a CAN bus off event, after 100 milliseconds. Feel free to adjust this if needed, but 100 milliseconds works fine in most cases.

With the can0 SocketCAN network interface configured, we can continue with the last step, which is bringing it in the UP state:

    sudo ip link set up can0

At this point you can run the command below to confirm the state
   ip -s -d link show can0
Result should be something like this (bold is mine):
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast **state UP** mode DEFAULT group default **qlen 1024**
    link/can  promiscuity 0  allmulti 0 minmtu 0 maxmtu 0
    can state ERROR-ACTIVE **restart-ms 100**
          **bitrate 1000000** sample-point 0.750
          tq 62 prop-seg 5 phase-seg1 6 phase-seg2 4 sjw 2 brp 3
          gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
          clock 48000000
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus usb parentdev 2-1.4:1.0
    RX:  bytes packets errors dropped  missed   mcast
             0       0      0       0       0       0
    TX:  bytes packets errors dropped carrier collsns
             0       0      0       0       0       0

_Note: If you restart, the CAN will be in **state "down."**To solve this, we need to take an additional step to automatically put state UP on reboot_

6. Automatically bring up the SocketCAN network interface
From the terminal run the following command to create this file and to edit it with the Nano terminal text editor
    sudo nano /etc/modules-load.d/can.conf

Next, enter the can and can_raw kernel modules each on their own line:
can
can_raw

exit and save changes

Before we move on with the configuration of our SocketCAN network interface, we need to first activate the systemd-networkd service. Run the following two commands from the terminal:

    sudo systemctl start systemd-networkd
    sudo systemctl enable systemd-networkd

This starts the systemd-networkd service and makes sure it also starts the next time your boot your Linux system.

To enable this configuration file for our can0 SocketCAN network interface, we just need to restart the systemd-networkd service:

    sudo systemctl restart systemd-networkd

Run the command below to confirm the state "UP"
   ip -s -d link show can0
(optional, "sudo reboot" then run the above command to confirm state is "UP"

7. That's it

Esoterical, 2024-11-14:
Appreciate the info. This network management issue has been known for a bit, but I don't have any CB1/BTT Pi to test with so I wasn't confident in adding any info without personally testing. 
I've only really seen this issue crop up a few times, and unfortunately each time we try to go down to an actual fix the user ends up jsut installing the latest CB1 image from BTT which 'fixes' it.

itsmeelertime, 2024-12-11:
Thank you so much for this information! I was banging my head against the wall trying to figure it out but loading those modules fixed it.

clembo1, 2024-12-11:
I'm glad it worked out for you.  I can confirm that everything works going along this path.

Esoterical, 2024-12-11:
@clembo1 For what it's worth, I did end up doing further testing.
A clean latest Ubuntu server image worked perfectly fine with the netplan instructions I have in my guide currently.
Interestingly, I could also remove the can.link and can.network files (which then "broke" the CAN interface again) but then installed ifupdown with a `sudo apt-get install ifupdown` and then added the /etc/network/interfaces-d/can0 file stuff and this also allowed the can0 interface to function as intended.

If having both the netplan can.link and can.network files *and* the /interfaces.d/can0 file with ifupdown installed it seemed the ifupdown config took precedence.

So I'm not certain what was going on with your specific setup, or if there are other distros out there that don't work with the netplan setup I suggest, but Ubuntu server definitely worked fine.

clembo1, 2024-12-15:
Thank you for the update.  It could be confined to the "Armbian" Debain Bookworm distro specified as CLI/Server.  I ran a clean build so I don't really know why there would be issues.  The main reason I pursed the Systemd route was due to Debain pushing netplan coupled with the future potential of no longer providing ifupdown...basically trying to future proof my setup.

I haven't tried the ubuntu server, but I might in the future; for now my goal is finishing the printer cfg.

Esoterical, 2024-12-15:
so it seems armbian *minimal* might use systemd-networkd, and I suspect the current instructions would work for it, but server and desktop versions look like they're using NetworkManager which might leave us in the same situation as the current RaspiOS release.

https://docs.armbian.com/User-Guide_Networking/

I'll do more digging and test with both a minimal armbian and a server armbian to figure out the best way forward. This might explain why yours is behaving "differently" to what I've seen so far. I expect not a lot of users are running armbian server image.

clembo1, 2024-12-16:
You nailed it.  I opted for the more recent Armbian CB1 - Bookworm CLI/Server distro because the BTT image has not been updated in quite a while and because of a youtube video with Armbian Bookworm as a suggested option for my M8P+CB1.  I also wasn't certain that a minimal image would have everything I needed (specifically networking, but it looks like minimal would have been the best path in hindsight).  As result, I ended up in a more unique spot then what others may commonly encounter.

At a minimum, I was trying to provide a path for others who might have gone the same route as me and spend hours of time banging a head against a wall because of the "distro/version/time of day/dev's choice in socks"...lol, so true.  

If there is a way to further simplify instructions...I think that's awesome!  Tbh, I probably did overkill on what is actually needed but I wanted to be 100% certain that everything would work from an OS perspective.

clembo1, 2024-12-16:
I'm with you on the Can/Can-raw modules.  Since it was my first time through and I was desperate, I went with the direction from the web-page I originally linked.  Just reread the page, it was from 2021 and was for "SocketCAN" and external usb-to-can adapters which are different than the integrated boards of today.  Probably not needed; I did not test to confirm one way or the other, though I'm betting you are correct in simply updating the files and enabling networkd.

Thank you for following this up.  I really appreciate it.

Esoterical, 2024-12-16:
OK, Just tested, fresh latest armbian *server* image for raspi 4, plugged in a CAN adapter (old Mellow SB2040 in usb-can-bridge mode), added the 10-can.link and 25-can.network files as per my current instructions, then `sudo systemctl enable systemd-networkd` then rebooted.
once it was back up `ip -s -d link show can0` showed a can0 interface in ERROR-ACTIVE state with the correct speed and queue length, exactly as I'd hoped.

The only curious thing was the armbian boot time was stuck at waiting for the systemd-networkd-wait-online.service for a couple of minutes before moving on, but this might be because I have no actual network connected to this pi (no ethernet or wifi). Once it "failed" and continued everything worked as normal.

So I think I'll change my docs around using systemd.networkd as default anyway, it seems like it's more universal and having just one setup should be less confusing.

Esoterical, 2024-12-16:
I've made the changes on the getting_started page, it's all systemd-networkd now (and changed the qlen to 128 but that is unrelated to the network service choice)

Esoterical, 2024-12-17:
and it turns out that TransmitQueueLength was only added to .link files in systemd version 248. A bunch of distros (the BTT armbian ones, and maybe bullseye-era RaspiOS) have 247 so it didn't work.
Changed the instructions to just use a udev .rules file directly, I imagine this will work for everyone.

clembo1, 2024-12-25:
@Esoterical Thank you! for all of the extra work.  I have really enjoyed working with you and greatly appreciate your diligence on getting the additional bits and pieces lined up right.

Side note. Should I go ahead and close this one out now?  I don't think there is anything further needed.

closed 2025-08-27 by clembo1