Esoterical/voron_canbus #84: CAN bus states DOWN during setup

SejeAnd, 2024-11-12:
Hello people :)
I  got to the klipper flashing part via katapult.

I flashed to an BTT octopus pro H723 board with the CAN adapter built in to the motherboard (so no U2C or anything in that sort is needed)

started from the beginning of the guide by making the can0 file an followed the guide through without any issues. However now when i push a "ip -s -d link show can0" command in the CLI it writes the CAN state as DOWN instead of the desired UP.

I have checked if I did everything right in the warning sign section and after what i see everything is configured correctly.

I hope you can help. Thanks in advance! :D
[image]

clembo1, 2024-11-13:
CAN is going to be down by default.  The network service has to be brought up.  If you go the route of ifupdown, you will need to install it and nettools.  

For more details, see here https://github.com/Esoterical/voron_canbus/issues/83
and here: https://canbus.esoterical.online/Getting_Started.html

Esoterical, 2024-11-14:
@clembo1 *most* people don't need to install anything. It's only the edge case armbian-based distros that have issues that your issue ticket covers.
edge-case in terms of people running klipper-based printers.

SejeAnd, 2024-11-14:
> CAN is going to be down by default. The network service has to be brought up. If you go the route of ifupdown, you will need to install it and nettools (if using Armbian or debian or ubuntu). If you go the systemd route, the github issue I submitted should "work" and will clarify that path.

> 
> For more details, see here #83 and here: https://canbus.esoterical.online/Getting_Started.html

I followed your  tutorial (it was great - good job :D), and it kind of fixed my issue. The can0 network is in the "UP" state now, even when rebooting. However the qlen value is only at 10 - not the 1024 its supposed to.
I have double checked, and i certainly configured the systemd-networkd (netplan) and ifupdown configs correctly. What could cause my issue, and how can i fix it?

Thank you in advance :D

Various screenshots:
[image]
[image]
[image]
[image]