Esoterical/voron_canbus #86: [Suggestion] Using a `.link` file for setting `txqueuelen`

CRTified, 2025-01-03:
I've just seen the recent change to the guide that uses `systemd-networkd` for configuration of the CAN interface via a `.network` file.

However, the current form uses udev to set the transmit queue length. As the guide now uses `systemd-networkd` anyway, wouldn't it make sense to use a [`.link`](https://www.freedesktop.org/software/systemd/man/latest/systemd.link.html) file for setting the [transmit queue length](https://www.freedesktop.org/software/systemd/man/latest/systemd.link.html#TransmitQueueLength=)?

I'm using this file for a while now:
```ini
[Match]
Type=can

[Link]
TransmitQueueLength=128
```

It might make sense to use a different `[Match]` configuration, but the systems of most people following your guide probably have exactly one CAN bus anyway. [`Path`](https://www.freedesktop.org/software/systemd/man/latest/systemd.link.html#Path=) might be a sensible alternative for matching, in my case it would be `Path=platform-fe3a0000.usb-usb-0:1:1.0` as shown by `networkctl status can0`:

```
# networkctl status can0
● 3: can0
               Link File: /etc/systemd/network/10-can.link
            Network File: /etc/systemd/network/can0.network
                   State: carrier (configured)
            Online state: online
                    Type: can
                    Kind: can
                    Path: platform-fe3a0000.usb-usb-0:1:1.0
                  Driver: gs_usb
                  Vendor: OpenMoko, Inc.
                   Model: Geschwister Schneider CAN adapter
                     MTU: 16
                   QDisc: pfifo_fast
Number of Queues (Tx/Rx): 1/1
       Activation Policy: up
     Required For Online: yes
```

If desired, I can write a PR.

Esoterical, 2025-01-03:
I previously used/tried a .link file when the systemd-networkd setup was just the secondary option (with ifupdown being the primary).
But I've now found that `TransmitQueueLength=` was only added as a .link option in systemd version 248, and a lot of still commonly used distros we see in 3d printing have an older version and so this config is ignored.

Instead of trying to get a user to update their systemd I found it's just easier to use a udev rule. It's more universal, and using the one-line commands to create/modify the udev file anyway it doesn't add any extra "effort" to users (especially novices)

CRTified, 2025-01-03:
> lot of still commonly used distros we see in 3d printing have an older version

That's a fair point. I didn't have that in mind!

Thanks for the feedback :)

closed 2025-01-03 by CRTified