Esoterical/voron_canbus #87: Common Mainboard Hardware SKR 3 EZ Processor Model Choice

3DPrintDemon, 2025-01-09:
Hi, thank you for this guide.

Please would you be able to add a second processor choice for people using this mainboard as they were supplied with different chips depending rev/model. The page states to set them processor model to `STM32H723`, however  I have version of this board here with `STM32H743` chips. 

All other settings are the same as your current page for the mainboard, just processor model can vary.

https://github.com/Esoterical/voron_canbus/tree/main/mainboard_flashing/common_hardware/BigTreeTech%20SKR-3%20EZ

Esoterical, 2025-01-09:
Thanks! I'll add a note about the 120r resistor.
Also, I can see that they mention the 723/743 versions on a product page, but not on their github or anything. Do you know if the pin selection/clock/bootloader options are the same between them and it's just the STM chip choice itself that is different?

3DPrintDemon, 2025-01-10:
You’re welcome, I’m glad I could help here with this. 

Yes it’s only mentioned on their product page, the version of the board’s manual I have shows the 743 setting only as a processor choice. 

I can confirm the all options are indeed exactly the same as per your existing published examples. 
Mine simply required the 743 setting changed only. 

The system is now up & running/printing with these settings & all commands in the guide followed.