Esoterical/voron_canbus #92: Network time sync broken (Pi OS)

blackICE19240, 2025-01-26:
**Pi CM4 emmc - Pi OS Lite - Using WiFi
Manta m5p**

---
Following the instructions on the "getting started" page (and only that page) - then rebooting, results in the default method of time sync not working. 

I've reflashed stock Pi OS lite to confirm this, I think the only changes I made prior were doing updates (apt update && apt full-upgrade), installing fish (a different shell) and switching to using a keyfile for SSH.

Disabling `systemd-networkd` and rebooting, will allow the sync to work again  - however - I believe enabling this was not the step that stopped it working. I rebooted between the first steps and checked if time sync was still working, and it was.

- Enabling/disabling `systemd-networkd-wait-online.service` does not seem to affect this.

`$ timedatectl`
>    Local time: Sun 2025-01-26 05:26:23 GMT
>    Universal time: Sun 2025-01-26 05:26:23 UTC
>    RTC time: n/a
>    Time zone: Europe/London (GMT, +0000)
>    System clock synchronized: no
>    NTP service: active
>    RTC in local TZ: no

`$ timedatectl timesync-status`
>     Server: n/a (n/a)
>     Poll interval: 0 (min: 32s; max 34min 8s)
>     Packet count: 0

**Work around:**
sudo apt install chrony
sudo systemctl enable chrony
sudo systemctl start chrony

(chrony is able to handle time sync instead)

blackICE19240, 2025-01-26:
Oh, and the guide is excellent!
Thank you very much :)

Esoterical, 2025-01-27:
hmm, you're not the first to report this:
https://github.com/Esoterical/voron_canbus/issues/91
but I've had a quick look at my two devices that I've been testing on with the systemd-networkd and both still had time sync, but also these system's *don't* currently have canbus connected.

I need to do more investigation, just don't have the equipment right this minute

blackICE19240, 2025-01-27:
Sorry, I guess I didn't put much effort into searching :P This is definitely a duplicate.
Mine also does not have the toolhead connected.

What devices and linux distros are you using? If it isn't Pi OS (lite), maybe it's something that has ntp installed by default. I would guess that might solve it. I will check that now.

blackICE19240, 2025-01-27:
Update: 
Installing the NTP package also makes the time sync again.

Could you check if you have either ntp or chrony installed by default? It might just be Pi OS doesn't - I noticed my desktop linux is using chrony by default.

Esoterical, 2025-01-27:
definitely piOS lite, just not the latest latest.
V0: `Linux raspberrypi 6.1.0-rpi7-rpi-v7 #1 SMP Raspbian 1:6.1.63-1+rpt1 (2023-11-24) armv7l GNU/Linux`
Trident: `Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

blackICE19240, 2025-01-27:
I just want to check you saw my last comment, I think we posted them at the same time :)
___
I've seen [this](https://docs.redhat.com/en/documentation/red_hat_enterprise_linux/7/html/system_administrators_guide/ch-configuring_ntp_using_the_chrony_suite#ch-Configuring_NTP_Using_the_chrony_Suite) -  Red hat says that chrony is basically always preferred.
___
P.S. I realise that I searched for this issue three days ago, and only got around to it today so that's why I didn't see the other one 😆

blackICE19240, 2025-01-27:
With a lot of back and fourth with chatGPT, 'we' seem to have got to the bottom of it!

For context, I am using WiFi. I will only paste a few bits:

"The output from `networkctl` ... reveals a critical insight into the problem."
"wlan0 (Wi-Fi): Marked as `unmanaged` but also `routable`, which suggests the Wi-Fi interface is functional but not being fully managed by `systemd-networkd`."
"When an interface is `unmanaged`, `systemd-networkd` doesn't configure it, which might prevent `systemd-timesyncd` from working properly (as it needs a managed, routable network interface to reach NTP servers)."

### Anyway, here's the fix:
Make some files so it manages eth0 and/or wlan0 too.
`echo -e "[Match]\nName=eth0\n\n[Network]\nDHCP=yes" | sudo tee /etc/systemd/network/10-eth0.network > /dev/null`
`echo -e "[Match]\nName=wlan0\n\n[Network]\nDHCP=yes" | sudo tee /etc/systemd/network/10-wlan0.network > /dev/null`

One for wifi and one for ethernet, but do both.

I haven't actually checked everything else still works as it should. I will tomorrow. I'm tired.

Good night :P

blackICE19240, 2025-01-27:
Nearly forgot to mention, 
`echo -e "[Match]\nName=can*\n\n[CAN]\nBitRate=1M" | sudo tee /etc/systemd/network/25-can.network > /dev/null`
was the upsetting line.

GP3DS, 2025-01-30:
I was finally able to release 2 machines to test.

After several restarts, this solution is functional (WIFI and CANBus):

`echo -e "[Match]\nName=eth0\n\n[Network]\nDHCP=yes" | sudo tee /etc/systemd/network/10-eth0.network > /dev/null`
`echo -e "[Match]\nName=wlan0\n\n[Network]\nDHCP=yes" | sudo tee /etc/systemd/network/10-wlan0.network > /dev/null`

Not having an RJ45 connection for the moment, I cannot guarantee efficiency via this connection :-)

Thank you @blackICE19240 for exploring the problem and finding the solutions ;-)
@Esoterical, is it possible to update the "Getting Started" procedure so that others are not in the same situation?

blackICE19240, 2025-01-30:
Thank you for checking it. 

It might be worth mentioning that I haven't tried it wired either. I expect it will work but I will test it at some point.

blackICE19240, 2025-02-09:
Thanks for letting us know.

Unfortunately I think only @Esoterical can add this to the guide, and he seems to have vanished 👻

thijstriemstra, 2025-02-09:
> he seems to have vanished 👻

a commit was made 3 hrs ago so not vanished fortunately.

blackICE19240, 2025-02-09:
Oh right. I don't know my way around github 🤷

Esoterical, 2025-02-09:
I'm still around, just haven't looked at this specific issue.
So, if I'm understanding it correctly, the "fix" is just creating valid systemd.networkd config files for wlan0 and eth0 but they are just going to null?
Does this also solve the 30 second wait-for-timeout issue? If it does then this would be nicer than just disabling the wait timeout service.

blackICE19240, 2025-02-10:
> Does this also solve the 30 second wait-for-timeout issue?

I don't think I've actually seen this problem yet so I don't know 🙃

> does this mean that all wlan/eth networking is being handled by systemd-networkd and not NetworkManager?

Yes. I assumed this was the expected behaviour? Are you sure your installation isn't doing this?

>  My concern is if this could cause problems on particularly custom/weird distros if that is the case....

Currently it's causing problems on stock, common distros - so I think it wants a note at least 😂

Esoterical, 2025-02-11:
I just did a test with a clean raspiOS legacy 32bit (the bullseye image) which doesn't seem to use networkmanager as the primary networking service/configurator.
Adding the systemd-networkd 10-wlan.network file resulted in the pi getting *two* IP addresses on teh wlan interface. 

[image]

I can't imagine that will lead to good things. And the old bullseye image is still very commonly used. Things like this is why I'm very hesitant to add the current "fix" to the guide.

And sure, I could add some more "if you are seeing this then try these steps" options but from past experience that just leads to confusion for readers.

Especially when the only broken thing is "no NTP". Not ideal, for sure, but doesn't stop anything from working. I think a different workaround would be better.

blackICE19240, 2025-02-11:
Once the time is significantly different from 'real' time, you can no longer install or update packages without manually correcting the time - which people won't even know needs doing.

thijstriemstra, 2025-02-13:
I reverted the changes suggested in esoterical's guide, and used the official updated klipper guide instead: https://github.com/Klipper3d/klipper/pull/6798

worked great.

thijstriemstra, 2025-02-15:
> I reverted the changes suggested in esoterical's guide, and used the official updated klipper guide instead: https://github.com/Klipper3d/klipper/pull/6798
> worked great.

Unfortunately I spoke too soon, and found a working fix [after a lot of fiddling](https://github.com/Klipper3d/klipper/pull/6798#issuecomment-2661081651):

Add following to `/etc/systemd/network/25-can.network`:

```
[Link]
RequiredForOnline=no
```

And system time sync works again.

blackICE19240, 2025-02-16:
Thanks. I'll try that instead at some point :)

Esoterical, 2025-03-28:
So, I don't know how I missed @thijstriemstra's suggestion, but it does seem that adding that RequiredForOnline=no does fix the time sync issue. 
I've updated the guide so the "one click" command to write the 25-can.network file includes this option 

```
echo -e "[Match]\nName=can*\n\n[CAN]\nBitRate=1M\nRestartSec=0.1s\n\n[Link]\nRequiredForOnline=no" | sudo tee /etc/systemd/network/25-can.network > /dev/null
```

[image]

I still don't know if this 100% works on older distros (or off-brand flavours like the CB1 or other all-in-one board distros) but I can't find anything in the manpages that indicate it shouldn't.

thijstriemstra, 2025-03-29:
@Esoterical you still left this section in there:

```
Unfortunately, this also enables the systemd-networkd-wait-online service which can cause your system to be "frozen" for around 2 minutes on boot as it waits for network interfaces to be online. If the problem was only the CAN interface there would be ways around this but as there are edge cases where the wait would still happen I've found it's best to just disable the wait-online servce entirely by running:

sudo systemctl disable systemd-networkd-wait-online.service

If disabling this service results in any negative impact on certain linux distros or setups please let me know via an [issue ticket](https://github.com/Esoterical/voron_canbus/issues).
```

which should be removed since we do not want to disable the service.

Esoterical, 2025-03-29:
@thijstriemstra I'd love to remove it, but have you found a better solution to the problem?
the `RequiredForOnline=no` doesn't affect the wait-online waiting issue.

thijstriemstra, 2025-10-07:
>  I'd love to remove it, but have you found a better solution to the problem? the RequiredForOnline=no doesn't affect the wait-online waiting issue.

You're right, I just noticed this as well, it has to be removed. cheers.

using:

```
Linux ender 6.12.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.47-1+rpt1~bookworm (2025-09-16) aarch64 GNU/Linux
```