Esoterical/voron_canbus #94: Issue with Python3

Mart-GitHub, 2025-02-06:
I'm following the Esoterical guide to set up Can on my Manta M8P to EBBSB2209 and I cant locate the python3-serial, has anyone come across this issue and have a solution, please help.

[image]

[image]

jjl40, 2025-03-03:
I have encountered the same problem with M4P, CB2.
tried uninstalling and reinstalling Python3-serial. The CB2 recommended OS is [CB2_Debian11_Klipper_kernel4.19_20250220.img.xz](https://github.com/bigtreetech/CB2/releases/download/V1.0.5/CB2_Debian11_Klipper_kernel4.19_20250220.img.xz).

Esoterical, 2025-03-27:
@Mart-GitHub a bit late, but I've added a note to try using pip3 to install pyserial if you see issues with the python3-serial.
this was the older method, and so older distros (especially CBx ones apparently) may still need it.
Note, I've only tested this on a single user so while I hope it's a wide spread solution I don't know for sure yet.

[image]