Esoterical/voron_canbus #98: PICAN not showing UUID

KA009onPC, 2025-03-29:
i was able to get the PICAN to show up on my printer and everything, and it was showing up as CAN1.. sadly i downloaded a USB to USB instead of CAN stuff while setting up my box turtle and now i cant get it to see the CAN1 at all and i cant get it to show the UUID... if i connect and discounted the PICAN nothing changes in the LSUSB

[image]
[image]

Esoterical, 2025-03-29:
@KA009onPC If you have a PICAN, then you *shouldn't* have the "Geschwister Schneider CAN adapter" showing up to lsusb.
That would imply you've flashed USB-CAN-Bridge mode to a board, maybe your mainboard.

And yes, my guide does tell you how to do that, but *only* if you are going to use that mainboard as your primary CAN adapter as well. If you were doing that then you wouldn't need the PiCAN Hat.

So if you are using a PiCAN HAT, then you should flash your mainboard with normal klipper USB (you can find instructions at <https://usb.esoterical.online>

Once you've done that the PiCAN will default to can0 instead of can1.

KA009onPC, 2025-03-30:
i think thats what i did by accident.. everything was working then i downloaded the wrong thing... ill try that and go from there

KA009onPC, 2025-03-30:
will that mess up my other CAN stuff though? my tool head board is running CAN as well?

KA009onPC, 2025-03-30:
what happened was i was just going through the process
 for the box turtle and uploading klipper to it, but then realized i was in the USB section, and thats when i lost the USB to CAN UUID for my PICAN

KA009onPC, 2025-03-30:
> [@KA009onPC](https://github.com/KA009onPC) If you have a PICAN, then you _shouldn't_ have the "Geschwister Schneider CAN adapter" showing up to lsusb. That would imply you've flashed USB-CAN-Bridge [...]

so i ended up doing this and its getting late and im currently stuck with it not wanting to load klipper now ( i haven't done this too many times so im forgetting what to do... im sorry... here is my klipper log as well if you can see anything. 

currently i have followed your usb for the mainboard.

thanks! 

[klippy (2).log](https://github.com/user-attachments/files/19524157/klippy.2.log)

KA009onPC, 2025-03-30:
[klippy (4).log](https://github.com/user-attachments/files/19526519/klippy.4.log)