Esoterical/voron_canbus #99: BTT CB2 not booting after "Network Service, CAN Speeds, and Transmit Queue Length" commands

hitman01560, 2025-03-30:
hello,

currently setting up a Manta M4P with CB2 module using the v3 beta image. 

I tried to install an ebb36 with u2c board so I followed the "getting started" part to set up the can interface.

I used the commands from the "Network Service, CAN Speeds, and Transmit Queue Length" and had no issues.

but after rebooting in the end, my CB2 doesn't boot anymore.

I'm totally noob in Linux and I'm desperate to find a solution to get my canbus working.

I'll try the old method with can0 file to compare if this works vs systemd-networkd.

cb2 image is based on bookworm if this can help.

any advice/feedback is welcome.thank you.

hitman01560, 2025-03-30:
repeated the steps a 2nd time, but before reinstalling KIAUH and klipper and seems my CB2 is rebooting now.

hitman01560, 2025-03-30:
and nope. my cb2 refuses to boot again after I put the printer on 24v to flash klipper on the toolhead...

Esoterical, 2025-03-30:
do you have a klipperscreen on this printer?

hitman01560, 2025-03-31:
@Esoterical no I didn't install klipper screen during my 6 tries... it happens too after a "sudo apt update" and "sudo apt upgrade" with their minimal image, so it seems it's not related to your process. I think the BTT image is not stable when upgrading the system or my CB2 has a bootloader problem... Thank you for this great guide btw.