Esoterical/voron_canbus #100: SUGGESTION: Create a third "Peripheral" category

thetic, 2025-04-27:
AFC-Lite, BTT Eddy, etc. are all currently listed as "Toolhead Hardware." Meanwhile boards from Isik's Tech like the Nevermore PCBs and the Birds' Nest are missing. I think these boards are neither mainboards or toolhead boards. A third category called something like peripherals would be better.

Esoterical, 2025-04-27:
I completely understand where you are coming from with a "not toolhead but not mainboard" section.
The reason I haven't done that is simply to try and make it less confusing.
AFC-Lite/Eddy/PITB, while not toolhead boards, are flashed *exactly* the same way as a toolhead board.
Whereas something like the Birds' nest (which I just haven't added, TBH I didn't realise there was a CAN Bridge version of it) would go under the "mainboard" section as it would be flashed as a mainboard.

In earlier iterations of this guide I tried to cover more options and explain the differences and stuff like that, but I found that just led to even more confusion. I hoped that with the current format, if you are flashing usb-can-bridge then the "click here to find your board" link will cover *all* boards for canbridge whether or not they are a true mainboard. Same with toolheads.

Adding a third peripherals section either means it's just an option in the sidebar with no hyperlinks in the actual guide, or in the guide I'll need to have a "click `mainboard` or `peripherals` to get the board settings" and that just doesn't feel as tidy to me.

thetic, 2025-04-27:
I can see where you're coming from. I bring this up because I was going to try to contribute pages for some of these missing boards but couldn't figure out where they would go.

Esoterical, 2025-04-27:
If it's canbridge, put it in the mainboard section.
if it's pure canbus, put it in the toolhead section.

if it's pure USB then it goes on <https://usb.esoterical.online> 😜

closed 2025-04-28 by thetic