Esoterical/voron_canbus #102: SUGGESTION: Inform users that a 32-bit Pi OS installation is recommended in Getting Started and Troubleshooting pages

JoshuaKabo, 2025-05-10:
I have found many sources (see below) which indicate that 64-bit OS installations on raspberry pis result in more consistent CAN communication errors.

When installing Mainsail OS, Raspberry Pi Imager defaults to a 64-bit installation, which is what I used for my Pi3B+. Despite carefully following the troubleshooting steps in the Esoterical voron_canbus guide, I encountered frequent lost communications, communication timeout, and buffer issues with my CAN setup running 64-bit Mainsail OS. These issues would occur within 2 hours of starting each print.

From my layman's intuition it makes sense to me that when the rest of the network uses 32-bit processors, (STM32's on toolhead and mainboard) running a 32-bit OS on the Pi would reduce risk of communication errors as there is less surface area for translation issues. That, or maybe there are software implementation issues in how CAN communication software was written for 64 bit systems.

_The following sources corroborate my finding that 32-bit installations are less error prone when communicating over CAN:_
https://klipper.discourse.group/t/psa-canbus-on-32-bit-vs-64-bit-pis-communication-timeout-errors/18336
https://www.teamfdm.com/forums/topic/2148-can-bus-comms-issue/?do=findComment&comment=14697
https://klipper.discourse.group/t/installing-klipper-with-kiauh/10734
https://forum.vorondesign.com/threads/2209-rp2040-missed-scheduling-of-next-digital-out-event.1732/post-12955
https://forum.vorondesign.com/threads/2209-rp2040-missed-scheduling-of-next-digital-out-event.1732/post-11797
https://forum.vorondesign.com/threads/2209-rp2040-missed-scheduling-of-next-digital-out-event.1732/post-12469

I didn't think 32 v 64-bit installations could be an issue, I assumed that I was struggling with the hardware limitations of the Pi3B+. It wasn't until I bought a Pi5 and went to set it up from scratch that I found the above forum posts.

If this is a valid concern, I would think it is best to inform users both **upfront**, and in the **troubleshooting steps**. Thanks!!

Esoterical, 2025-05-11:
Thanks for the suggestion.

There *is* a bit of nuance to this issue. It's not so much 32 vs 64bit, I know for a *fact* that Klipper and canbus (or just any multi-mcu setup, the problem isn't CAN itself) runs perfectly fine on 64 bit systems.

The issue comes from the bullseye kernel (6.1.x IIRC) and 64bit. This absolutely causes weird timing issues. The reason this pops up to people as "64bit doesn't work" is because a lot of people just use MainsailOS as their image, and MainsailOS still uses the bullseye kernel.

If you use just vanilla "Raspberry Pi OS" in the imager then it will use the latest kernel (bookworm, 6.6.x or something) which I know works perfectly fine whether 32 or 64 bit.

But you are correct that this should be noted somewhere in the guide, it's something I haven't got around to doing yet.

Esoterical, 2025-05-12:
@JoshuaKabo I've added a section on the main "getting started" page about it https://canbus.esoterical.online/Getting_Started.html#raspberry-pi-operating-system

and I've also added a new troubleshooting section of "other errors" which goes into some things and one of them redirects back to this front page.
I'll probably also add a reference back to it in the TTC troubleshooting page.

JoshuaKabo, 2025-05-12:
> [@JoshuaKabo](https://github.com/JoshuaKabo) I've added a section on the main "getting started" page about it https://canbus.esoterical.online/Getting_Started.html#raspberry-pi-operating-system
>
> and [...]

Perfect! Thank you!

closed 2025-05-12 by JoshuaKabo