Esoterical/voron_canbus #104: MainsailOS 2.0.0 CAN Bus Setup - NetworkManager and Queue Length Fixes

Dario503, 2025-05-31:
## Summary
This documents two issues found with CAN bus setup on MainsailOS 2.0.0 and their solutions:
1. NetworkManager interference with CAN interfaces
2. Race condition causing incorrect queue length settings

> [!IMPORTANT]
> NetworkManager conflicts with systemd-networkd for CAN interface management

> [!WARNING]  
> The default queue length of 10 is unexpectedly set in spite of 10-can.rules content

Setup:

Raspberry Pi Imager MainsailOS 2.0.0 - Raspberry Pi  (64-bit) image Released 2025-05-19 ships with ONLY NetworkManager installed and enabled. 

Hardware under test: 

RPi5, BTT Octopus Pro, BTT SB2209 CAN toolhead board

Specific OS details:

```bash
$ uname -a
Linux voron2 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

systemd-networkd: inactive
networking:              inactive
dhcpcd:                    inactive
NetworkManager:   active
connman:                 inactive

The instructions here: [Esoterical](https://canbus.esoterical.online/Getting_Started.html#network-service-can-speeds-and-transmit-queue-length
) enable systemd-networkd for managing can interfaces. This is fine but there can be conflicts with Network Manager. It is important to tell NetworkManager to ignore can*:

```bash
sudo mkdir -p /etc/NetworkManager/conf.d
cat <<EOF | sudo tee /etc/NetworkManager/conf.d/10-unmanaged-can0.conf
[keyfile]
unmanaged-devices=interface-name:can*
EOF
sudo systemctl reload NetworkManager
```
I discovered another issue with the queue-length being set for the default: 10 even though the 10-can.rules file was setup up per the Esoterical document. The root cause was that the USB-CAN adapter (via the gs_usb driver) often creates and “ups” the can0 interface faster than networkd can configure it—so the kernel default queue (10) sticks. By also setting up a udev rule, the queue length is set to 128 the moment a can  interface appears, even before networkd runs. This can be done as follows:

```bash
cat <<EOF | sudo tee /etc/udev/rules.d/10-can-queue.rules
SUBSYSTEM=="net", ACTION=="add|change", KERNEL=="can*", ATTR{tx_queue_len}="128"
EOF
sudo udevadm control --reload
sudo udevadm trigger --subsystem-match=net --action=change
```
With these 2 additions, the Esoterical instructions result in a working system with the 5-19-2025 Mainsail image.

Esoterical, 2025-05-31:
Maybe this is a pi5 specific thing (though that would be surprising).
After seeing this I did a test on a brand new flash of a CM4 on a manta. Used Raspi imager, flashed the latest mainsailOS, SSH'd in and just followed the getting_started page to enable systemd-networkd, disabled the systemd-networkd-wait-online service, created the 10-can.rules file (using the `echo -e` string), added the 25-can.network file (again using the `echo-e` string) and then rebooted.

The manta had been previously flashed with usb-can-bridge mode so I didn't need to reflash that at all.

When it came back up the can0 interface was showing with the 128qlen as expected.

[image]

[image]

Dario503, 2025-05-31:
Pi4 works fine for me as well. Pi5 build is where it showed up. I embarked on the switch to pi5 because the pi4 would occasionally lock up on long prints. I suspected it was usb related; I'm using a c920 usb camera and usb to Octopus. The pi5 has different usb architecture. So far so good but it has its own challenges with low frame rates with usb cameras (well documented). When I get some time, I'll test this on another pi5 and see if the qlen problem shows up.

Esoterical, 2025-05-31:
I just did a fresh mainsailOS 2.0 install to my pi5 (with a kraken in usb-can-bridge mode) and once again only booted the pi, enabled systemd-networkd, disabled systemd-networkd-wait-online, then set the 10-can.rules and 25-can.network then rebooted.
the can0 interface shows as expected with 128 qlen

[image]

What device are you using for your CAN interface?

Dario503, 2025-05-31:
Octopus Pro v1.0 with USB to can bridge klipper v0.12. Can cable to btt
2209 tool head board.

For the pi image I used menu selection from RPi imager. Rpi5 -> Other
specific OS, 3d printing, Mainsail, Mainsail 2.0.0 rpi 64-bit

As opposed to using kiauh

I’m in the middle of a long test print, I’ll find some time and wipe the
pi5 and reinstall.

Esoterical, 2025-05-31:
> Here's another approach to avoid dealing network management tools using NetworkManager exclusively to manage CAN as well with some additional info re: pi5
> 
> [Setting up pi5 / CAN with NetworkManager](https://github.com/JamesH1978/KlipperStuff/blob/main/Pi5CAN0.md)
> 
> Also, potentially someone else struggling with this:
> 
> [qlen=10 inspire of config](https://www.reddit.com/r/klippers/comments/1gtl9lh/qlen_value_issues_in_canbus_config/)

Some background on these things as well,
The first one *definitely* needs systemd-networkd enabled in order to function (the `/etc/systemd/network/80-can0.network` is a config file for systemd-networkd, and this service is installed but not enabled by default on new distros).

Something that would affect both of the pages you linked is the old method of using the `10-can.link` file. It turns out that the  ```
[Link]
TransmitQueueLength=128
```
option wasn't available in older systemd versions, which a lot of people were/are still using. This is why I changed to using a straight udev rule for the queuelen (you can probably find the issue ticket about it somewhere here)

Dario503, 2025-05-31:
Thanks, I really do appreciate your well thought out guide to Canbus!