Esoterical/voron_canbus #114: Broken install while upgrading to 6.12.47 Debian bookworm (2025-09-16) aarch64 GNU/Linux

theogayar, 2025-09-24:
As stated in the title, I broke my install on a debian RPI4 (ran for months) by updating to the latest stable release. I'm opening this for future reference.

The issue seems to be that the upstream modified somehow the way networkd handled canbus, adding a check for hardware support before enabling the interface. This results in an error message when the daemon tries to enable the can0 interface, and it simply returns "Device doesn't support..." 

I believe this will never be fixed upstream, as it is not a bug, but rather a feature, altough I cannot figure out why someone would need this.

So the fix is : 

Remove the old config file :
`sudo rm /etc/systemd/network/25-can.network`

Create a new systemd service: 
`sudo nano /etc/systemd/system/can0.service`

Paste inside that service : 
```
[Unit]Description=CAN interface can0After=network.target

[Service]
Type=oneshot
ExecStart=/sbin/ip link set can0 up type can bitrate 1000000
ExecStop=/sbin/ip link set can0 down
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

Enable and start the service:

```
sudo systemctl daemon-reload
sudo systemctl enable can0.service
sudo systemctl start can0.service
```

For new installs, before starting the service, also run : 

`echo -e 'SUBSYSTEM=="net", ACTION=="change|add", KERNEL=="can*"  ATTR{tx_queue_len}="128"' | sudo tee /etc/udev/rules.d/10-can.rules > /dev/null
`

I hope this will help someone!

Esoterical, 2025-09-24:
@theogayar 
I just had a user (on the voron discord) have issues with this, so was able to realtime test things.

They were on  `6.12.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.47-1+rpt1~bookworm (2025-09-16) aarch64 GNU/Linux` and with the `RestartSec=0.1s` in the config their can0 network wouldn't auto start. The interface would be there but show as "down" when doing an `ip -s -d link show can0`

If they ran `sudo ip link set can0 up type can bitrate 1000000` it would bring the interface online which showed that the interface *did* work, it just wasn't auto starting.

I got them to simply remove the `RestartSec=0.1s` line from the `/etc/systemd/network/25-can.network` file and reboot and after that everything worked as intended. The can0 interface would start up on boot and also after a klipper firmware_restart.

I'm going to modify the instructions to omit the RestartSec option. I don't know why it's a problem now, the default behavior is 100ms restart delay anyway so really "RestartSec=0.1s" shouldn't be changing anything. Maybe the latest debian/kernel/whatever has a new systemd version that just doesn't like RestartSec. It needs further investigation.

When I first changed the instructions over to systemd.networkd (from ifupdown) I didn't have restartsec in the config at all and things still worked. I think someone said to add it to help with when a CAN device is being flakey to stop the can0 interface from just erroring out to a `down` state but I may be misremembering. Need to dig through issue tickets.

@Zaraton I've tried searching through systemd documentation and I can't find mention of `restart-ms` being a valid config option at all. So either it is and I just can't find the documentation page for it, or it isn't valid and it's just ignoring that and the reason it "fixed" it for you was just because you removed `RestartSec`. Not sure.

theogayar, 2025-09-25:
Just for the record, and replying to @Zaraton : 

To fix the restarts using a service, I did this : 

Script in /usr/local/bin/can0-up-watch.sh (don't forget `sudo chmod +x /usr/local/bin/can0-watch.sh`)
```
#!/bin/bash

while true; do

    # Bring up can0 only if it's down
    if ! ip link show can0 &> /dev/null || ! ip link show can0 | grep -q "UP"; then
        echo "Bringing up can0..."
        ip link set can0 up type can bitrate 1000000
    fi

    # Wait a bit before checking again
    sleep 5
done
```

And updated the service accordingly: in /etc/systemd/system/can0.service
```
[Unit]
Description=Persistent CAN bus can0 for Klipper
After=network.target
Wants=network.target

[Service]
Type=simple
ExecStart=/usr/local/bin/can0-up-watch.sh
Restart=always
RestartSec=2
StartLimitIntervalSec=0

[Install]
WantedBy=multi-user.target
```

Now to @Esoterical , I had noticed exactly the same as you, the interface was indeed there and down (see the script, I was checking specifically if it was up, not just if it was there). I just removed the `RestartSec`, and tried setting everything back up the way it was previously, and it works 👍 

Not exactly sure why the behavior changed, but there is indeed also an issue with the current man page for networkd : 

> RestartSec= Automatic restart delay time. If set to a non-zero value, a restart of the CAN controller will be triggered automatically in case of a bus-off condition after the specified delay time. Subsecond delays can be specified using decimals (e.g. "0.1s") or a "ms" or "us" postfix. Using "infinity" or "0" will turn the automatic restart off. **By default, automatic restart is disabled.** 

Which is untrue, as I could verify just now, automatic restart is enabled by default, and almost instantaneous. 

TL:DR : 

Comment out/remove `RestartSec=0.1s` from the can config file at `/etc/systemd/network/25-can.network` then reboot. 

@Esoterical, I'll let you close when you are done with the edits. Thanks for this guide, it really is nice to have around!

thijstriemstra, 2025-09-30:
> Comment out/remove RestartSec=0.1s from the can config file at /etc/systemd/network/25-can.network then reboot.

Thanks removing that line also fixed it for me after updating RPI earlier today:

```
Linux foobar 6.12.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.47-1+rpt1~bookworm (2025-09-16) aarch64 GNU/Linux
```

Esoterical, 2025-10-01:
Requiredforonline was found to be needed in order for NTP time sync to function. Without it then (for whatever reason) the OS would be waiting for systemd-networkd to handle NTP which it obviously wouldn't do over canbus and generally the eth0/wlan0 isn't handled by networkd either.
So setting requiredforonline=no in the can interface stopped the OS from using networkd for NTP entirely.

No idea of the underlying shenanigans for it but that was the reason.